IOsetting/FwLib_STC8 #10: Question regarding PWM DAC example

AnonymousChatter, 2022-12-30:
I have been trying out the "pwm_dac_voice" example from the Demo folder on a STC8H8K64U chip, not too sure if I am doing anything wrong but I am just getting screeching noise from pin P1.0 instead of the "fire" alert. 

Would anyone have any advice on where I should start looking?

Thanks :)

IOsetting, 2023-01-01:
I just did a test on a 48-pin STC8H8K64U, it works, the output pin is P1.0.

One possible problem is the speaker resistance, if it is too low, the chip may not be able to drive it. Add a 100R resistor between P1.0 and speaker will fix this.

Another possible problem might be the frequency, add some code to print through uart. If the frequency is incorrect, the output will be garbled.
```c
void main(void)
{
    SYS_SetClock();
    PWM_Init();
    Timer0_Init();
    UART1_Config8bitUart(UART1_BaudSource_Timer1, HAL_State_ON, 115200);

    while(1)
    {
        UART1_TxString("some string\r\n");
        SYS_Delay(1000);
    }
}
```

So far the frequency cannot be adjusted by stc8prog [Feat(firc calibration): feature request](https://github.com/IOsetting/stc8prog/issues/8), you need to manually set the registers to reach the target frequency in the code.

The env of my STC8H8K64U in platformio.ini is
```
; ======================== stc8h8k64u ========================
;           STC8H8K64U   ␍␊
;CPUID:                  F784C9090000A7␍␊
;Vref 1.19V:             0x04A3␍␊
;32KHz:                  0x86C4␍␊
;22.1184MHz:             0x36␍␊
;24MHz:                  0x55␍␊
;20MHz       27MHz:      0x8D␍␊
;27MHz       30MHz:      0xC2␍␊
;30MHz       33.1176MHz: 0xFE␍␊
;33.1776MHz  35MHz:      0x4C␍␊
;35MHz       36.864MHz:  0x66␍␊
;36.864MHz   40MHz:      0x7E␍␊
;40MHz       44.2368MHz: 0xAC␍␊
;45MHz       48MHz:      0xD2␍␊
;VRTRIM 6MHz:            0x1F␍␊
;VRTRIM 10MHz:           0x22␍␊
;VRTRIM 27MHz:           0x1E␍␊
;VRTRIM 44MHz:           0x1B␍␊
;
; 24MHz: Current IRCBAND:82, VRTRIM:1E, IRTRIM:55, LIRTRIM:A8
; 36M864 Current IRCBAND:83, VRTRIM:1B, IRTRIM:5E, LIRTRIM:A8
[env:stc8h8k64u]
platform = intel_mcs51
board = STC8H8K64U
build_flags =
    -D__CONF_FOSC=44236800UL
    -D__CONF_MCU_MODEL=MCU_MODEL_STC8H8K64U
    -D__CONF_CLKDIV=0x00
    -D__CONF_IRCBAND=0x03
    -D__CONF_VRTRIM=0x1B
    -D__CONF_IRTRIM=0xAC
    -D__CONF_LIRTRIM=0x01
upload_protocol = custom
upload_port = /dev/ttyUSB0
upload_speed = 115200
upload_flags =
    -p$UPLOAD_PORT
    -s$UPLOAD_SPEED
    -e
upload_command = ${platformio.packages_dir}/tool-stc8prog/stc8prog $UPLOAD_FLAGS -f $SOURCE
```
44.23680MHz is reached by the following configs: 
* `__CONF_CLKDIV=0x00`, no division
* `__CONF_IRCBAND=0x03`, stc8h8k64u has 4 IRC bands(0x00 - 0x03), 44MHz is the highest one so it is 0x03
* `__CONF_VRTRIM=0x1B`, the factory-set of `VRTRIM 44MHz` is 0x1B
* `__CONF_IRTRIM=0xAC`,  the factory-set of `44.2368MHz` is 0xAC
* `__CONF_LIRTRIM=0x01`, this value 

If you use STC-ISP, it will be much easier -- set `-D__CONF_FOSC` to the frequency, and comment out `SYS_SetClock();` from the code, then use STC-ISP to set the frequency when flashing.

AnonymousChatter, 2023-01-01:
Thank you so very much!!! ~> Your build flag works flawlessly.
A noob mistake, I simply set the "-D__CONF_FOSC" hoping that it will set up the frequency correctly in the chip. (Good way to test with your suggested UART method).

closed 2023-01-01 by AnonymousChatter