IOsetting/FwLib_STC8 #11: Thanks and a fix

porcinus, 2023-02-11:
First, a huge thanks for you library.
I never used a STC8 MCU before and your work greatly helped me :)

I made a "helper" file to compute and reverse compute some values related to prescalers.
If you are interested, you can borrow the file or any parts of it ;)
https://github.com/porcinus/NNS-Ambiclone-Mono/blob/main/code/src/FwLib_STC8_helper.h

During coding, I also notice an error (I think, haven't tried but this generate a error) in https://github.com/IOsetting/FwLib_STC8/blob/master/include/fw_rcc.h
In macro RCC_SetPowerDownWakeupTimerCountdown(__16BIT_COUNT__):
(__16BIT_COUNT__ << 8) should be (__16BIT_COUNT__ >> 8)

IOsetting, 2023-02-12:
Thanks for the report, there are quite some errors in fw_rcc.h. I have updated it accordingly. Please let me know if there are any other errors.

IOsetting, 2023-02-13:
FwLib_STC8_helper.h looks nice, I will borrow some code if you don't mind :smile:

porcinus, 2023-02-13:
> Thanks for the report, there are quite some errors in fw_rcc.h. I have updated it accordingly. Please let me know if there are any other errors.

This was the only error i noticed :)

porcinus, 2023-02-13:
> FwLib_STC8_helper.h looks nice, I will borrow some code if you don't mind smile

Take any parts you want ;)