IOsetting/FwLib_STC8 #12: Keil编译环境下使用定时器1作为串口波特率发生器报错

perseverance51, 2023-03-06:
在uart1_timer1_tx.c文件中原外部变量引入：

`extern __CODE uint16_t ticks_ms;
extern __CODE uint8_t  ticks_us, clkdiv;`

需要如下修改：
`
#if defined (SDCC) || defined (__SDCC)
extern __CODE uint16_t ticks_ms;
extern __CODE uint8_t  ticks_us;                //clkdiv (include fw_reg_stc8h.h)
#elif defined __CX51__
extern uint16_t code  ticks_ms;
extern uint8_t  code ticks_us;
#endif

……

 UART1_TxHex(CLKDIV);//改成大写

`
在`fw_sys.c`文件中调整如下：
`#if defined (SDCC) || defined (__SDCC)
__CODE uint16_t ticks_ms    = (__CONF_FOSC / ((__CONF_CLKDIV == 0)? 1 : __CONF_CLKDIV) / __CLK_REF);
__CODE uint8_t  ticks_us    = (__CONF_FOSC / ((__CONF_CLKDIV == 0)? 1 : __CONF_CLKDIV) / __CLK_REF / 1000);
#elif defined __CX51__
uint16_t code  ticks_ms    = (__CONF_FOSC / ((__CONF_CLKDIV == 0)? 1 : __CONF_CLKDIV) / __CLK_REF);
uint8_t  code  ticks_us    = (__CONF_FOSC / ((__CONF_CLKDIV == 0)? 1 : __CONF_CLKDIV) / __CLK_REF / 1000);
#endif`

IOsetting, 2023-03-07:
Thank you for the report. The code should work as `__CODE ` macro will be translated to `code` in `__CX51__`, which is defined in fw_reg_base.h
```c
#elif defined __CX51__
...
    #define __CODE  code
    #define __REENTRANT  reentrant
```
I will test it in Keil C51 later.

IOsetting, 2023-03-12:
The commit e2024f59fb70bef0d531be6a2b3c550f6a7a98e8 has fixed this issue.