IOsetting/FwLib_STC8 #17: Multiple definition error when using stdint.h

eeyrw, 2023-06-20:
Those new C51 compiler such as SDCC supports stdint.h which contains platform independent definition such as int8_t, uint8_t 8,etc. However the FwLib_STC8 defines same things at fw_types.h, which finally make a lot errors when compiling the code using stdint.h.  Could we rewrite the part of fw_types.h like this:
```C
.............
#if defined (SDCC) || defined (__SDCC)
#include <stdint.h>
#else
typedef unsigned char uint8_t;
typedef unsigned short uint16_t;
typedef unsigned long uint32_t;
typedef uint32_t uint64_t[2];

typedef signed char int8_t;
typedef short int16_t;
typedef long int32_t;
typedef int32_t int64_t[2];
#end
............
```

IOsetting, 2023-06-20:
Sure, please create a PR for this change, I will be happy to merge this.

eeyrw, 2023-06-20:
https://github.com/IOsetting/FwLib_STC8/pull/18

closed 2023-06-20 by eeyrw