IOsetting/FwLib_STC8 #21: PlatformIO编译问题

dasudiy, 2024-08-16:
include "fw_hal.h"后编译空的main()报错

- MCU: STC8G1K08
- OS: Windows11 23H2
- VSCODE: 1.92.2
- PlatformIO: Core 6.1.15

# platformio.ini

```ini
[env:STC8G1K08]
platform = intel_mcs51
board = STC8G1K08
build_flags =    
    -D__SDCC_SYNTAX_FIX
    -D__CONF_FOSC=26000000UL
    -D__CONF_MCU_MODEL=MCU_MODEL_STC8G1K08
    -D__CONF_IRCBAND=0x03
    -D__CONF_VRTRIM=0x19
    -D__CONF_IRTRIM=0x28
    -D__CONF_LIRTRIM=0x00
```

# main.c
```c
#include "fw_hal.h"

int main(void)
{
    return 0;
}

```

# terminal output
```shell
 *  Executing task: C:\Users\Asuka\.platformio\penv\Scripts\platformio.exe run --verbose --environment STC8G1K08 

Processing STC8G1K08 (platform: intel_mcs51; board: STC8G1K08; build_flags: -D__SDCC_SYNTAX_FIX, -D__CONF_FOSC=26000000UL, -D__CONF_MCU_MODEL=MCU_MODEL_STC8G1K08, -D__CONF_IRCBAND=0x03, -D__CONF_VRTRIM=0x19, -D__CProcessing STC8G1K08 (platform: intel_mcs51; board: STC8G1K08; build_flags: -D__SDCC_SYNTAX_FIX, -D__CONF_FOSC=26000000UL, -D__CONF_MCU_MODEL=MCU_MODEL_STC8G1K08, -D__CONF_IRCBAND=0x03, -D__CONF_VRTRIM=0x19, -D__CONF_IRTRIM=0x28, -D__CONF_LIRTRIM=0x00)
-----------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/intel_mcs51/STC8G1K08.html
PLATFORM: Intel MCS-51 (8051) (2.2.0) > Generic STC8G1K08
HARDWARE: STC8G1K08 11MHz, 1.25KB RAM, 8KB Flash
PACKAGES: 
 - toolchain-sdcc @ 1.40400.0 (4.4.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- FwLib_STC8 @ 1.0 (License: Unknown, Path: D:\APX20XInit\lib\FwLib_STC8)
Building in release mode
sdcc -o .pio\build\STC8G1K08\src\main.rel -c --std-sdcc11 --opt-code-size --peep-return -mmcs51 -DF_CPU=11059200L -DHEAP_SIZE=128 -DPLATFORMIO=60115 -DSTC8G1KXX -DSTC8G1K08 -DNAKED_ARCH_MCS51 -DNAKED_MCS51_STC8G1KXX -D__SDCC_SYNTAX_FIX -D__CONF_FOSC=26000000UL -D__CONF_MCU_MODEL=MCU_MODEL_STC8G1K08 -D__CONF_IRCBAND=0x03 -D__CONF_VRTRIM=0x19 -D__CONF_IRTRIM=0x28 -D__CONF_LIRTRIM=0x00 -Iinclude -Isrc -Ilib\FwLib_STC8\include -Ilib\FwLib_STC8\src src\main.c
sdcc -o .pio\build\STC8G1K08\libcdb\FwLib_STC8\fw_adc.rel -c --std-sdcc11 --opt-code-size --peep-return -mmcs51 -DF_CPU=11059200L -DHEAP_SIZE=128 -DPLATFORMIO=60115 -DSTC8G1KXX -DSTC8G1K08 -DNAKED_ARCH_MCS51 -DNAKED_MCS51_STC8G1KXX -D__SDCC_SYNTAX_FIX -D__CONF_FOSC=26000000UL -D__CONF_MCU_MODEL=MCU_MODEL_STC8G1K08 -D__CONF_IRCBAND=0x03 -D__CONF_VRTRIM=0x19 -D__CONF_IRTRIM=0x28 -D__CONF_LIRTRIM=0x00 -Ilib\FwLib_STC8\include -Ilib\FwLib_STC8\src lib\FwLib_STC8\src\fw_adc.c
sdcc -o .pio\build\STC8G1K08\libcdb\FwLib_STC8\fw_i2c.rel -c --std-sdcc11 --opt-code-size --peep-return -mmcs51 -DF_CPU=11059200L -DHEAP_SIZE=128 -DPLATFORMIO=60115 -DSTC8G1KXX -DSTC8G1K08 -DNAKED_ARCH_MCS51 -DNAKED_MCS51_STC8G1KXX -D__SDCC_SYNTAX_FIX -D__CONF_FOSC=26000000UL -D__CONF_MCU_MODEL=MCU_MODEL_STC8G1K08 -D__CONF_IRCBAND=0x03 -D__CONF_VRTRIM=0x19 -D__CONF_IRTRIM=0x28 -D__CONF_LIRTRIM=0x00 -Ilib\FwLib_STC8\include -Ilib\FwLib_STC8\src lib\FwLib_STC8\src\fw_i2c.c
sdcc -o .pio\build\STC8G1K08\libcdb\FwLib_STC8\fw_mem.rel -c --std-sdcc11 --opt-code-size --peep-return -mmcs51 -DF_CPU=11059200L -DHEAP_SIZE=128 -DPLATFORMIO=60115 -DSTC8G1KXX -DSTC8G1K08 -DNAKED_ARCH_MCS51 -DNAKED_MCS51_STC8G1KXX -D__SDCC_SYNTAX_FIX -D__CONF_FOSC=26000000UL -D__CONF_MCU_MODEL=MCU_MODEL_STC8G1K08 -D__CONF_IRCBAND=0x03 -D__CONF_VRTRIM=0x19 -D__CONF_IRTRIM=0x28 -D__CONF_LIRTRIM=0x00 -Ilib\FwLib_STC8\include -Ilib\FwLib_STC8\src lib\FwLib_STC8\src\fw_mem.c
sdcc -o .pio\build\STC8G1K08\libcdb\FwLib_STC8\fw_spi.rel -c --std-sdcc11 --opt-code-size --peep-return -mmcs51 -DF_CPU=11059200L -DHEAP_SIZE=128 -DPLATFORMIO=60115 -DSTC8G1KXX -DSTC8G1K08 -DNAKED_ARCH_MCS51 -DNAKED_MCS51_STC8G1KXX -D__SDCC_SYNTAX_FIX -D__CONF_FOSC=26000000UL -D__CONF_MCU_MODEL=MCU_MODEL_STC8G1K08 -D__CONF_IRCBAND=0x03 -D__CONF_VRTRIM=0x19 -D__CONF_IRTRIM=0x28 -D__CONF_LIRTRIM=0x00 -Ilib\FwLib_STC8\include -Ilib\FwLib_STC8\src lib\FwLib_STC8\src\fw_spi.c
sdcc -o .pio\build\STC8G1K08\libcdb\FwLib_STC8\fw_sys.rel -c --std-sdcc11 --opt-code-size --peep-return -mmcs51 -DF_CPU=11059200L -DHEAP_SIZE=128 -DPLATFORMIO=60115 -DSTC8G1KXX -DSTC8G1K08 -DNAKED_ARCH_MCS51 -DNAKED_MCS51_STC8G1KXX -D__SDCC_SYNTAX_FIX -D__CONF_FOSC=26000000UL -D__CONF_MCU_MODEL=MCU_MODEL_STC8G1K08 -D__CONF_IRCBAND=0x03 -D__CONF_VRTRIM=0x19 -D__CONF_IRTRIM=0x28 -D__CONF_LIRTRIM=0x00 -Ilib\FwLib_STC8\include -Ilib\FwLib_STC8\src lib\FwLib_STC8\src\fw_sys.c
sdcc -o .pio\build\STC8G1K08\libcdb\FwLib_STC8\fw_tim.rel -c --std-sdcc11 --opt-code-size --peep-return -mmcs51 -DF_CPU=11059200L -DHEAP_SIZE=128 -DPLATFORMIO=60115 -DSTC8G1KXX -DSTC8G1K08 -DNAKED_ARCH_MCS51 -DNAKED_MCS51_STC8G1KXX -D__SDCC_SYNTAX_FIX -D__CONF_FOSC=26000000UL -D__CONF_MCU_MODEL=MCU_MODEL_STC8G1K08 -D__CONF_IRCBAND=0x03 -D__CONF_VRTRIM=0x19 -D__CONF_IRTRIM=0x28 -D__CONF_LIRTRIM=0x00 -Ilib\FwLib_STC8\include -Ilib\FwLib_STC8\src lib\FwLib_STC8\src\fw_tim.c
sdcc -o .pio\build\STC8G1K08\libcdb\FwLib_STC8\fw_uart.rel -c --std-sdcc11 --opt-code-size --peep-return -mmcs51 -DF_CPU=11059200L -DHEAP_SIZE=128 -DPLATFORMIO=60115 -DSTC8G1KXX -DSTC8G1K08 -DNAKED_ARCH_MCS51 -DNAKED_MCS51_STC8G1KXX -D__SDCC_SYNTAX_FIX -D__CONF_FOSC=26000000UL -D__CONF_MCU_MODEL=MCU_MODEL_STC8G1K08 -D__CONF_IRCBAND=0x03 -D__CONF_VRTRIM=0x19 -D__CONF_IRTRIM=0x28 -D__CONF_LIRTRIM=0x00 -Ilib\FwLib_STC8\include -Ilib\FwLib_STC8\src lib\FwLib_STC8\src\fw_uart.c
In file included from lib\FwLib_STC8\include/fw_reg_stc8g.h:4,
                 from lib\FwLib_STC8\include/fw_conf.h:155,
                 from lib\FwLib_STC8\include/fw_hal.h:18,
                 from src\main.c:19:
lib\FwLib_STC8\include/fw_reg_base.h:21:7: warning: #warning unrecognized compiler
   21 |     # warning unrecognized compiler
      |       ^~~~~~~
In file included from lib\FwLib_STC8\include/fw_reg_stc8g.h:4,
                 from lib\FwLib_STC8\include/fw_conf.h:155,
                 from lib\FwLib_STC8\include/fw_adc.h:18,
[... 196 lines cut ...]
  to type 'volatile-unsigned-char generic* fixed'
lib\FwLib_STC8\src\fw_i2c.c:97: warning 88: cast of LITERAL value to 'generic' pointer
from type 'const-unsigned-int literal'
  to type 'volatile-unsigned-char generic* fixed'
lib\FwLib_STC8\src\fw_i2c.c:100: warning 88: cast of LITERAL value to 'generic' pointer
from type 'const-unsigned-int literal'
  to type 'volatile-unsigned-char generic* fixed'
lib\FwLib_STC8\src\fw_i2c.c:101: warning 88: cast of LITERAL value to 'generic' pointer
from type 'const-unsigned-int literal'
  to type 'volatile-unsigned-char generic* fixed'
lib\FwLib_STC8\src\fw_i2c.c:104: warning 88: cast of LITERAL value to 'generic' pointer
from type 'const-unsigned-int literal'
  to type 'volatile-unsigned-char generic* fixed'
lib\FwLib_STC8\src\fw_i2c.c:108: warning 88: cast of LITERAL value to 'generic' pointer
from type 'const-unsigned-int literal'
  to type 'volatile-unsigned-char generic* fixed'
lib\FwLib_STC8\src\fw_i2c.c:111: warning 88: cast of LITERAL value to 'generic' pointer
from type 'const-unsigned-int literal'
  to type 'volatile-unsigned-char generic* fixed'
*** [.pio\build\STC8G1K08\libcdb\libFwLib_STC8.lib] Error 3221225595
============================================================================================ [FAILED] Took 4.63 seconds ============================================================================================

 *  The terminal process "C:\Users\Asuka\.platformio\penv\Scripts\platformio.exe 'run', '--verbose', '--environment', 'STC8G1K08'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

dasudiy, 2024-08-16:
I also tried this on ubuntu and got more detailed logs. It seems I don't have enough ram for it. So I'd like to know if this project is capable with STC8G1K08 or not. Any help will be appreciated.

```shell
 *  Executing task: platformio run --verbose --environment STC8G1K08 

Processing STC8G1K08 (platform: intel_mcs51; board: STC8G1K08; build_flags: -D__SDCC_SYNTAX_FIX, -D__CONF_FOSC=26000000UL, -D__CONF_MCU_MODEL=MCU_MODEL_STC8G1K08, -D__CONF_IRCBAND=0x03, -D__CONF_VRTRIM=0x19, -D__CONF_IRTRIM=0x28, -D__CONF_LIRTRIM=0x00)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/intel_mcs51/STC8G1K08.html
PLATFORM: Intel MCS-51 (8051) (2.2.0) > Generic STC8G1K08
HARDWARE: STC8G1K08 11MHz, 1.25KB RAM, 8KB Flash
PACKAGES: 
 - toolchain-sdcc @ 1.40100.12072 (4.1.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- FwLib_STC8 @ 1.0 (License: Unknown, Path: /home/asuka/Documents/PlatformIO/Projects/HelloWorld/lib/FwLib_STC8)
Building in release mode
sdcc -o .pio/build/STC8G1K08/src/main.rel -c --std-sdcc11 --opt-code-size --peep-return -mmcs51 -DF_CPU=11059200L -DHEAP_SIZE=128 -DPLATFORMIO=60115 -DSTC8G1KXX -DSTC8G1K08 -DNAKED_ARCH_MCS51 -DNAKED_MCS51_STC8G1KXX -D__SDCC_SYNTAX_FIX -D__CONF_FOSC=26000000UL -D__CONF_MCU_MODEL=MCU_MODEL_STC8G1K08 -D__CONF_IRCBAND=0x03 -D__CONF_VRTRIM=0x19 -D__CONF_IRTRIM=0x28 -D__CONF_LIRTRIM=0x00 -Iinclude -Isrc -Ilib/FwLib_STC8/include -Ilib/FwLib_STC8/src src/main.c
In file included from lib/FwLib_STC8/include/fw_reg_stc8g.h:4,
                 from lib/FwLib_STC8/include/fw_conf.h:155,
                 from lib/FwLib_STC8/include/fw_hal.h:18,
                 from src/main.c:1:
lib/FwLib_STC8/include/fw_reg_base.h:21:7: warning: #warning unrecognized compiler
sdcc -o .pio/build/STC8G1K08/libd91/FwLib_STC8/fw_adc.rel -c --std-sdcc11 --opt-code-size --peep-return -mmcs51 -DF_CPU=11059200L -DHEAP_SIZE=128 -DPLATFORMIO=60115 -DSTC8G1KXX -DSTC8G1K08 -DNAKED_ARCH_MCS51 -DNAKED_MCS51_STC8G1KXX -D__SDCC_SYNTAX_FIX -D__CONF_FOSC=26000000UL -D__CONF_MCU_MODEL=MCU_MODEL_STC8G1K08 -D__CONF_IRCBAND=0x03 -D__CONF_VRTRIM=0x19 -D__CONF_IRTRIM=0x28 -D__CONF_LIRTRIM=0x00 -Ilib/FwLib_STC8/include -Ilib/FwLib_STC8/src lib/FwLib_STC8/src/fw_adc.c
sdcc -o .pio/build/STC8G1K08/libd91/FwLib_STC8/fw_i2c.rel -c --std-sdcc11 --opt-code-size --peep-return -mmcs51 -DF_CPU=11059200L -DHEAP_SIZE=128 -DPLATFORMIO=60115 -DSTC8G1KXX -DSTC8G1K08 -DNAKED_ARCH_MCS51 -DNAKED_MCS51_STC8G1KXX -D__SDCC_SYNTAX_FIX -D__CONF_FOSC=26000000UL -D__CONF_MCU_MODEL=MCU_MODEL_STC8G1K08 -D__CONF_IRCBAND=0x03 -D__CONF_VRTRIM=0x19 -D__CONF_IRTRIIn file included from lib/FwLib_STC8/include/fw_reg_stc8g.h:4,
M=0x28 -D__CONF_LIRTRIM=0x00 -Ilib/FwLib                 from lib/FwLib_STC8/include/fw_conf.h:155,
                 from lib/FwLib_STC8/include/fw_adc.h:18,
                 from lib/FwLib_STC8/src/fw_adc.c:15:
lib/FwLib_STC8/include/fw_reg_base.h:21:7: warning: #warning unrecognized compiler
_STC8/include -Ilib/FwLib_STC8/src lib/FwLib_STC8/src/fw_i2c.c
In file included from lib/FwLib_STC8/include/fw_reg_stc8g.h:4,
                 from lib/FwLib_STC8/include/fw_conf.h:155,
sdcc -o .pio/build/STC8G1K08/libd91/FwLib_STC8/fw_mem.rel -c --std-sdcc11 --opt-code-size --peep-return -mmcs51 -DF_CPU=11059200L -DHEAP_SIZE=128 -DPLATFORMIO=60115 -DSTC8G1KXX -DSTC8G1K08 -DNAKED_ARCH_MCS51 -DNAKED_MCS51_STC8G1KXX -D__SDCC_SYNTAX_FIX -D__CONF_FOSC=260000                 from lib/FwLib_STC8/include/fw_i2c.h:18,
00U                 from lib/FwLib_STC8/src/fw_i2c.c:15:
Llib/FwLib_STC8/include/fw_reg_base.h:21:7: warning: #warning unrecognized compiler
 -D__CONF_MCU_MODEL=MCU_MODEL_STC8G1K08 -D__CONF_IRCBAND=0x03 -D__CONF_VRTRIM=0x19 -D__CONF_IRTRIM=0x28 -D__CONF_LIRTRIM=0x00 -Ilib/FwLib_STC8/include -Ilib/FwLib_STC8/src lib/FwLib_STC8/src/fw_mem.c
lib/FwLib_STC8/src/fw_i2c.c:21: warning 88: cast of LITERAL value to 'generic' pointer
from type 'const-unsigned-int literal'
  to type 'volatile-unsigned-char generic* fixed'
lib/FwLib_STC8/src/fw_i2c.c:22: warning 88: cast of LITERAL value to 'generic' pointer
from type 'const-unsigned-int literal'
  to type 'volatile-unsigned-char generic* fixed'
lib/FwLib_STC8/src/fw_i2c.c:23: warning 88: cast of LITERAL value to 'generic' pointer
from type 'const-unsigned-int literal'
  to type 'volatile-unsigned-char generic* fixed'
lib/FwLib_STC8/src/fw_i2c.c:24: warning 88: cast of LITERAL value to 'generic' pointer
from type 'const-unsigned-int literal'
  to type 'volatile-unsigned-char generic* fixed'
lib/FwLib_STC8/src/fw_i2c.c:25: warning 88: cast of LITERAL value to 'generic' pointer
from type 'const-unsigned-int literal'
  to type 'volatile-unsigned-char generic* fixed'
lib/FwLib_STC8/src/fw_i2c.c:28: warning 88: cast of LITERAL value to 'generic' pointer
from type 'const-unsigned-int literal'
  to type 'volatile-unsigned-char generic* fixed'
lib/FwLib_STC8/src/fw_i2c.c:29: warning 88: cast of LITERAL value to 'generic' pointer
from type 'const-unsigned-int literal'
  to type 'volatile-unsigned-char generic* fixed'
[... cut ...]
*** [.pio/build/STC8G1K08/firmware.hex] Error 1
=================================================================================================================================== [FAILED] Took 1.08 seconds ===================================================================================================================================

 *  The terminal process "platformio 'run', '--verbose', '--environment', 'STC8G1K08'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

600MHz, 2024-08-20:
You should not add "-D__SDCC_SYNTAX_FIX" to the flags for building environment as document mentioned that. (Read the English Manual rather than Chinese version)

BTW the building issue for my model `STC8G1K08A` also bothered me and I found that the build_flags in the comment of file `/demo/spi/max7219/max7219_8x8led_stc8g1k08.c` in this repo works fine. You can try it out.

dasudiy, 2024-08-21:
> You should not add "-D__SDCC_SYNTAX_FIX" to the flags for building environment as document mentioned that. (Read the English Manual rather than Chinese version)
>
> BTW the building issue for my model [...]

Thanks for the advice. I gave the build_flags from the demo code you mentioned a shot, but didn't have any luck. I'm still getting the same error on Windows. I'll try it on Ubuntu later.

Mobilluk, 2024-11-03:
I also have the same error:
```
Processing stc8g1k08 (platform: intel_mcs51; board: stc8g1k08)
----------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/intel_mcs51/stc8g1k08.html
PLATFORM: Intel MCS-51 (8051) (2.2.0) > Generic STC8G1K08
HARDWARE: STC8G1K08 11MHz, 1.25KB RAM, 8KB Flash
PACKAGES:
 - toolchain-sdcc @ 1.40400.0 (4.4.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- FwLib_STC8
Building in release mode
Compiling .pio\build\stc8g1k08\src\main.rel
Compiling .pio\build\stc8g1k08\lib637\FwLib_STC8\fw_adc.rel
Compiling .pio\build\stc8g1k08\lib637\FwLib_STC8\fw_i2c.rel
Compiling .pio\build\stc8g1k08\lib637\FwLib_STC8\fw_mem.rel
Compiling .pio\build\stc8g1k08\lib637\FwLib_STC8\fw_spi.rel
Compiling .pio\build\stc8g1k08\lib637\FwLib_STC8\fw_sys.rel
Compiling .pio\build\stc8g1k08\lib637\FwLib_STC8\fw_tim.rel
Compiling .pio\build\stc8g1k08\lib637\FwLib_STC8\fw_uart.rel
Compiling .pio\build\stc8g1k08\lib637\FwLib_STC8\fw_usb.rel
Compiling .pio\build\stc8g1k08\lib637\FwLib_STC8\fw_util.rel
Archiving .pio\build\stc8g1k08\lib637\libFwLib_STC8.lib
*** [.pio\build\stc8g1k08\lib637\libFwLib_STC8.lib] Error 3221225781
=================================================== [FAILED] Took 5.03 seconds ===================================================
 *  The terminal process "C:\Users\User\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

zyjwdrrr, 2025-03-25:
哥们，别听他们胡说。
删掉-D__SDCC_SYNTAX_FIX后那些warning会消失。
最后的报错是sdcc没有完整安装导致的，看issue 24